ostaquet/Arduino-MQ131-driver #3: question

rookettk, 2018-12-14:
Hi!, Thank u very much  for your library. I had done  with it good but I have a problem: Time to read of the sensor is take a long time, it spent  72s for time to read sensor. Can you help me fix it, i want get  less time to read sensor. Thanks

ostaquet, 2018-12-26:
Hi,
The time required to read the value is depending on the sensor. This kind of sensor is working by burning gas with the heather and measuring the concentration of some specific particles after a specific heating time.

The default heating time is 72s with the low concentration version and 80s with the high concentration version.

It is defined in the constructor of the driver:
`// Setup default calibration value
  switch(model) {
    case LOW_CONCENTRATION :
      setR0(110470.60);
      setTimeToRead(72);
      break;
    case HIGH_CONCENTRATION :
      setR0(385.40);
      setTimeToRead(80);
      break;
  }`

Those default values have been defined by experiments on the device I own. 

The "time to read" should be the delay between the start of the heater and the moment when the value is stable. If you decide to measure the value of the device after 5 seconds, the value will be erroneous. The best way to define the correct value is to use the method `calibrate()`.

After the calibration, you are able to read the value defined by the calibration with the method `getTimeToRead()` and you can change manually this heating time with the method `setTimeToRead(long sec)`.

Kind regards,
Olivier

MartinPellizzer, 2019-06-29:
Hi, I greatly appreciate your library and the work you did with the MQ131, because I have serious problems making it works.

I tried the default diagram given by the manufacturer, but the ADC readings shift too much during time and I can't have a stable reading, so I'm looking forward your library.

I tested your library and used your wiring diagram, but I'm having 1 issue. I can pass the calibration process and get 0.01ppm in clean air, but when I blow ozone to the sensor the reading doesn't change. I exposed the sensor to a lot of ozone for hours, but no results. Am I missing something? The sensor doesn't get hot like it does with the manufacturer diagram, but it stays almost cold. Is it normal behaviour?

Your support would be a life-save for me.

Kind regards,
Martin