ostaquet/Arduino-MQ131-driver #4: PPB goes DOWN in this library when PPB goes UP in real life

OldNewbold, 2019-03-06:
Just put my low concentration sensor in front of an ozone generator, was sitting at 12ppb, quickly dropped to 4ppb.

The datasheet for this product states "The sensor's conductivity gets lower along with the gas concentration rising" 

So we should expect the RS/R0 ratio to DECREASE as the ozone increases. (R0 is a constant)
e.g.                       Rs/R0 = 250k/200k = 1.25 
Gas increases       Rs/R0 = 220k/200k = 1.1

As you can see, as gas concentration increases, Rs/Ro should decrease. 

But the datasheet shows a graph of the Rs/R0 ratio INCREASING as the PPB rises!
It blatantly contradicts itself.
So long story short but your library literally works backwards for me.
Have you tested this working on a low concentration sensor with an ozone generator?

OldNewbold, 2019-03-06:
P.S. Ozone sensing seems to be dominated by untested products. Very Troubling.

ostaquet, 2019-03-09:
Dear OIdnewbold,
I didn't have the chance to test the sensor in a lab environment. I tested it by using the sensor outside in clear air and compare with results from a local air quality station.

According to the specification, the _sensor conductivity gets lower along with the gas concentration rising_. It means that the resistance (Rs) increase while the gas concentration is higher. To be accurate, it's important that the R0 is correctly calibrated. In my experience, I saw crazy output if the preheat period of 48 hours is not respected.

So, we have Rs increasing. So, Rs/R0 is increasing and according to the sensitivity curves, it gives us the concentration based on the formula defined in the [Excel file](Arduino-MQ131-driver/datasheet/Sensitivity_curves.xlsx).

I honestly don't see why your sensor is working backwards... except if it's not a MQ131 (a lot of Winsen sensors, like the MQ7, are working on the other way: if gas concentration is increasing, the resistance is becoming lower).

closed 2019-03-21 by ostaquet