ostaquet/Arduino-MQ131-driver #6: Heater minimum current

tukno, 2019-06-29:
Hello, I have a question related to the minimum consumed current of atleast 150mA. I checked the datasheet and it does not mention the value for the current, but if we assume the sensor is supplied 5V and consumes the maximum heater consumption of 900mW, using P = V*I, the minimum current should have been 180mA, right?

ostaquet, 2019-07-05:
Hi Tukno, your calculation is correct. The minimum current drained through the heater is approx. 180mA. The point in the explanation is that the Ardiuno pins support max 150mA, that's why you should never connect the heater directly on the Arduino pins.

closed 2019-07-05 by ostaquet

ferreiradiego, 2020-06-28:
Hello, So Is it better use a external supply for the heater (when the sensor is reading) or after the preheat time (approximately 48 hours) I can connect directly on the Arduino?

SciWax, 2020-06-28:
> Hello, So Is it better use a external supply for the heater (when the sensor is reading) or after the preheat time (approximately 48 hours) I can connect directly on the Arduino?

Hey diego. Unfortunately not. The heater of the sensor Always has to be connected to an external power supply. The MQ131 sensors (low and high concentration variants) consume just too much current for an Arduino to handle. You Need a separate 5V power supply, which can deliver enough current for the sensor. you can connect the sensor Pins directly to the Arduino though.

ferreiradiego, 2020-06-28:
Thanks @SciWax for your answer.

So, should I only connect the heater pins to an external power supply? the others pins (reading pins and Vcc) can I connect directly to the Arduino?

SciWax, 2020-06-28:
The datasheet tells you, that the heater  needs 5V DC or AC and the Vc-Pins also need 5V. Personally, I've connected Vh and Vc to my power supply, because my power supply provides 5V and enough current for the Heater and the sensor itself. You can connect the reading Pins to the Arduino, but don't Forget the load resistance. The datasheet and the circuit diagram of ostaquet give you a good idea About how to connect the sensor pins to your Arduino.

SciWax, 2020-07-14:
> Thanks @SciWax for your answer.
> 
> So, should I only connect the heater pins to an external power supply? the others pins (reading pins and Vcc) can I connect directly to the Arduino?

BTW, something I've forgot to mention. Be aware of the logic level of your Arduino. I'm using a Mega2560, which is 5V. According to the datasheet, the sensor can give you outputs around 4V, which is too much for the ADC of a 3,3V Arduino. Use a logic level converter, if you use a 3,3V Arduino!