ostaquet/Arduino-MQ131-driver #8: No readable output from Sensor

CAMR94, 2019-07-05:
Hello,
I was trying to implement the Ozone Sensor in a project I'm working on.
I imported the library and used the exact example code. 
Also I recreated the circuit with the bakelite low concentration version.
So With all of that it should work...but it doesnt't. 
The Output is as follows:

Callibration in Progress...
?F?

and after that it just doesn't proceed with anything.
I don't really know, what I am doing wrong.
Any help would be greatly appreciated.

Greetings from Germany.
Chris

ostaquet, 2019-07-05:
Hi Chris,
Seems strange... The line _Calibration in progress..._ is in the main function and should be followed by the _Calibration done!_ in the example. The calibrate() method doesn't output anything so I don't understand why you get the _?F?_.
Which Arduino board are you using?
Did you check the sensor with a multimeter?
Did you change something in the code?
KR,
Olivier

CAMR94, 2019-07-05:
Hi Oliver,

Thank you for the reply.

I did not change anything in the Code:

I'm using a Sainsmart Knockoff of the Arduino Uno. 

By now I actually don't even get the Calibration in Progress message. Only the weird ?F?.

I checked the circuit. The Heater circuit is getting 6.5V and the Sensor circuit 3.5V

Greetings, Chris

ostaquet, 2019-07-05:
Hi Cris,
Did you configure correctly the baud rate of the serial debug interface?
In the sketch embedded in the example, the serial is configured at 115200bps.
KR,
Olivier

CAMR94, 2019-07-09:
Hello Oliver,
You were right, the Baud rate was wrong. Thank you very much.
I am now running it at 9600bps and I'm getting a reasonable output. But the Sensor shows a saturation of "inf ppm" etc. so every value is at infinity. But I had no time yet to let it burn in, so I guess I will do that next.

Again thank you very much.
Greetings,
Chris

closed 2019-07-10 by ostaquet