ostaquet/Arduino-MQ131-driver #9: Sensor showing Rs of inf Ohms and not heating up

CAMR94, 2019-07-25:
Hello,

After my last Problem I was again working on the Ozone Sensor.

I am not sure what's going on, but if I start the program the calibration is finished under a minute and either shows 0 Ohms, or inf Ohms as the R0 value.

Also the Sensor does not feel like it is warming up.

I measured the Voltages: the heater circuit gets 5V and the Sensor circuit about 2.6V

I am really sure that my wiring is correct. I use the same setup as you and have bought the same Moduls(Transistor is a bcc33725, since I saw a few different versions there)

Do you have any clue for me as to what might be wrong? Even without the "burn in" it should heat up and give me some results at least shouldn't it?

closed 2019-08-01 by CAMR94

ostaquet, 2019-08-02:
Hi,

I just coming back from annual leave and see your issue. This behaviour is abnormal. If the heater receives 5V, it should heating up. Maybe you're experiencing hardware issue with your sensor.

I see that you closed the issue yesterday. How do you solve it? (it could help others ;-) )

KR

CAMR94, 2019-08-02:
It was a mistake on my part. I had misjudged the Pin Layout of the Sensor(Or at least that's what I think, I just wired it up from scratch again and it worked and I think I switched around some Pins on the Sensor)

ostaquet, 2019-08-02:
OK, thanks for sharing ;-)