ostaquet/Arduino-MQ131-driver #11: calibration time

CAMR94, 2019-08-05:
One Question. How long does the calibration usually take? My Sensors heat up proberly and seem fine, but the calibration has been running for half the day now and nothing happened. Is it calibrating for the whole 48h?

ostaquet, 2019-08-06:
The calibration process should be quite rapid (max 10 minutes).

The calibration process as built in the library works by heating the sensor for 15-20 seconds (depending on the sensor type) and measuring the value every seconds. At each cycle, we stored the value read on the sensor. As soon as the value is stable, we know the stabilization cycle of the sensor. It defined the R0 and the time to read.

Did you preheat your sensor? (burn time of 48h)

CAMR94, 2019-08-06:
I have not burned it in completely yet. it is running right now with the calibration in the background and it should be finished tomorrow morning. 

Is that burn in time necessary only once, or every time, that you are using it?

ostaquet, 2019-08-06:
Burn in time is required only once.
You should rerun the calibration after the burn in time.
If you have still problem, we have to investigate why the reading value is unstable.

CAMR94, 2019-08-07:
So I left it running and started the calibration again, after the burn in time. 
Still the calibration is not finishing. 

I added some Pictures of my Setup. It's a mess, but maybe you can see something, that I can't. 
Thank you again for your help.
[image]
[image]
[image]

ostaquet, 2019-08-08:
I didn't check in details your wiring. I have just some doubts about the power. It seems that you power everything through the USB of the computer, which is maybe not enough. I personally use the additional jack power to give enough power to the wiring.

In order to understand what is happening, you should add this after the line 246 in the MQ131.cpp file :
`Serial.println(value);`

It will give you an idea of the readings on the analog input.

CAMR94, 2019-08-14:
Hello again. Sorry for the late answer, I was gone for a few days.

I did add the jack, but still the Calibration doesn't finish.
I also added the Line and I am getting some readouts now. 

Here are some Images of the Readout at the start and roughly 10-15 minutes later.

[image]
[image]

ostaquet, 2019-08-23:
The variance is important and unusual. When I looked to your last output, we have the same values coming out : `1127777.75`, `1014000.00` and `920909.12`. During the calibration process, the code is expecting a stable value.

What is the board that you're using?
What is the load resistance you're using?

CAMR94, 2019-08-23:
https://ardubotics.eu/en/microcontrollers/150-sainsmart-uno-r3-atmega328p-development-board-compatible-with-arduino-uno-r3.html This is the Board that I'm using.

I have replicated your Setup exactly, so my resistors are the same. I set it up multiple times with the same results.

ostaquet, 2019-08-28:
I retest the whole code and made some technical refactoring. I don't see why your sensor is driving crazy during the calibration process.
Please check the connections as described in the README (I added the MQ131 pinout on a schema).
If you used a wrong connection on the pinouts, you had maybe damage the lead in the sensor. :-(

atlask1, 2019-09-03:
Hi  @CAMR94 ,
it's easy that you have damage the sensor, maybe you have powered it with wrong wiring. If you read the datasheet you could understand it. It's a supposition , but could happen.

[image]

this is extremely situation, but strange behaviour happen if it's partially damaged, like unstable value.

closed 2019-09-24 by ostaquet