ostaquet/Arduino-MQ131-driver #14: about calibrate function

flaviomo, 2020-05-08:
I'm using the driver but I'm apparently having problems with the "calibrate" function, it's taking too long to get out of the function, can you explain me why.
 I'm using an ozone generator to test the driver. 
Thank you in advance

CarlosFonseca97, 2020-05-10:
Gola buen dia, estuve analizando el tiempo y tiene que ver con el precalentamiento del sensor que generalmente se recomienda sea entre 24H-48H. Para la función MQ131.sample(); el tiempo que tarda en tomar una lectura dura 80 segundos y es para estabilizar el valor de medición

ostaquet, 2020-05-23:
@CarlosFonseca97 is right.

Before the first use, you have to "preheat" the sensor. To do this, just put 5V on the heater for 48 hours (use a power supply because batteries will be quickly empty).

closed 2020-05-23 by ostaquet