ostaquet/Arduino-MQ131-driver #17: Default values in MQ131.h

chazbevo, 2020-05-30:
Can I modify the default values in the #define statements? Default RL to the resistor value I have chosen?  I have a MQ131 module and there's a 110 ohm resistor on the board for RL.

I plan on having a temp/humidity sensor in a vessel.  Can I take those readings and place them in the default values of the #define statements?

Where did you get the default Base Resistance R0 for the high concentration sensor of 235.0?

Thanks,

ostaquet, 2020-05-31:
The `#define` statements in the file `MQ131.h` are default values to ensure that the library is running out of the box. They can, of course, be modified in the source code but it is not the best way to do it.

### To define the load resistance (RL)

The initialization of the driver is done with the function `begin()`. This function takes important data in the parameters like:
- the power pin to enable/disable the heater
- the analog pin connected to the sensor 
- the model of MQ131 (low concentration or high concentration)
- the load resistance value in Ohms

In the example [calibrate_and_read.ino](https://github.com/ostaquet/Arduino-MQ131-driver/blob/master/examples/calibrate_and_read/calibrate_and_read.ino):
```
  // Init the sensor
  // - Heater control on pin 2
  // - Sensor analog read on pin A0
  // - Model LOW_CONCENTRATION
  // - Load resistance RL of 1MOhms (1000000 Ohms)
  MQ131.begin(2,A0, LOW_CONCENTRATION, 1000000);  
```

The RL value set in the initialisation of the driver overrides the default value set in the `#define` statements.

### To use the correct temperature and humidity

In order to increase the accuracy of the concentration values, it is important to inform the driver about the temperature and the humidity level.

Those values can be set with the `setEnv()` function. This function has to be called before using the `getO3()` function.

Per example:
```
// Set environment conditions at 17°C and 35% of humidity
setEnv(17, 35);
```

Those environmental values override the default values set in the `#define` statements.

### To set the base resistance R0 with calibration

The R0 mentioned in the `#define` statements works for my sensors and each sensor is different due to the nature and the manufacturing process.

The R0 is given by the driver after the calibration process. The calibration process can be triggered by the function `calibrate()`.

**Warning: Before running the calibration process, don't forget to pre-heat the sensor for 48 hours.**

By running the example [calibrate_and_read.ino](https://github.com/ostaquet/Arduino-MQ131-driver/blob/master/examples/calibrate_and_read/calibrate_and_read.ino), the output on the Serial will be something like this:

```
Calibration in progress...
Calibration done!
R0 = 1917.22 Ohms
Time to heat = 80 s
Sampling...
Concentration O3 : 0.01 ppm
Concentration O3 : 7.95 ppb
Concentration O3 : 0.02 mg/m3
Concentration O3 : 16.80 ug/m3
```

This is how to get the R0 value. The R0 value can be set by using the function `setR0()`.

chazbevo, 2020-05-31:
Thank you for the prompt response.  From what I read:
1) the R0 value is unique to each sensor.  I ran the calibrate_and_read.ino several times.  The R0 value settled in around 11 ohms.  [seems low compared to yours] 

So, I can change the default value to 11 ohms or use the setR0() function?

2) Rather than change the "MQ131.h" default values, your recommendation is to set values prior to the initialization.  Would this be done within the read_high_concentration.ino sketch?  If I read your response correctly, I inserted the following within the setup(), AFTER the Serial.begin(), and BEFORE the MQ131.begin() initialization:
-  MQ131.setEnv(25,51);
-  MQ131.setR0(11);

Is this the correct syntax and placement of the code to set the variables?

Thanks again!

ostaquet, 2020-06-01:
Your R0 value is low compared to mine because you’re using a low RL. According to your first comment, you’re using a load resistance of 110Ohms. The datasheet recommends a load resistance of 1MOhms. My experience shows also good results with 10KOhms.

 I suggest you read carefully the source code in order to avoid issues 😉 

As you can see, all values are initialized in the `begin()` function. It means that all changes should be done after the call to the `begin()` function.

You can indeed change the R0 value through the `setR0()` function and set the environmental conditions through the `setEnv()` function but you have to do it **after** the initialization of the drive.

closed 2020-06-03 by ostaquet

chazbevo, 2020-07-01:
I got the "HI.CONCENTRATION" ino to work and have good readings.  I'm trying to get this sensor and a MHT11 temp/humidity sensor to operate on the same sketch and output T / H / O3 on the same display.  Baud rate on MQ131 sketch is 115200, on DHT11, the baud rate is 9600.  Can I either change the MQ131 sketch to  9600 or operate two baud rates on separate serial ports?  If I do that, I don't think I can monitor both.  Thanks!

toze3, 2020-07-30:
After testing alot this circuit and finding some problems with the very high impedance of the sensor (low concentration) I'm getting R0 of 250kohms, and this is very near the ones in the datasheet. Please check Figure 5 you can see the output voltage at zero ppb is around 4V and he uses a RL of 1Mohm, so by the ohm low and thinking as a voltage divider the resistance R0 of the sensor should be RL / (RL + R0) -> R0=250Kohm to get the Vrl of 4V (Vcc =5V) . Also form the some Figure for the concentration of 200 ppb you have a Vrl of 2.5V this means Rs=RL, and by this you have Rs/R0=4 for 200ppb now to confirm this calculation just check Figure 3 and see Rs/R0 is equal to 4 in 200ppb. I do believe there are differences between sensors, but it's hard to believe there is a diference in a factor of 100 or more. This is more or less I got from the LOW capacity from winsen.
just my two cents. 
note: for the high concentration the only difference should be the R0 used and the unit.