ostaquet/Arduino-MQ131-driver #18: About the influence of temperature and humidity in the values.

engdiego, 2020-06-02:
In your study, do you realized that the temperature and humidity influence a lot in the concentration values of  O3?

I did that question, because I am studying this sensor MQ131 (Low Concentration) and seems to me that the temperature and humidity do not influence a lot in the values.

Congratulations for your code, it's excellent

ostaquet, 2020-06-03:
The temperature and humidity influence has been designed based on [data sheet details](https://github.com/ostaquet/Arduino-MQ131-driver/blob/master/extras/datasheet/MQ131-low-concentration.pdf) on page 4.

The formulas have been designed based on the [sensitivity curves](https://github.com/ostaquet/Arduino-MQ131-driver/blob/master/extras/datasheet/Sensitivity_curves.xlsx).

ostaquet, 2020-06-18:
Additional comment on this, @nitram147 fixed some bugs in the calculation of the environmental correction. Please use the v1.4.0 to have the last version.

ferreiradiego, 2020-06-28:
Hello @nitram147.  Could You explain to me why the value of resistance is divided by the value of  `getEnvCorrectRatio()`? As in the line code `correctedValueR0 = _valueR0 / getEnvCorrectRatio();`. My doubt is about how the resistance ratio is corrected, is the code to correct the ratio only a mathematical artifice?

PS.: I'm doing a study about MQ Sensors and the more details I get, the better will be my study.

ostaquet, 2020-07-07:
The idea is to compare apples with apples. ;-)

According to the data sheet, the resistance of the sensor is affected by the temperature and humidity. This impact is implemented with the corrective ratio coming from `getEnvCorrectRatio()`.

But... this should be in the `calibrate()` function and not in the `setR0()`.

ostaquet, 2020-07-07:
It is quite a good question...

At the moment, the `correctedValueR0` is defined when the `setR0()` function is called. According to me, it is not exactly correct.

The R0 should be adjusted when we calibrate the sensor and when we get the R0 value (because the R0 value evolved with the environment).

ostaquet, 2020-07-07:
I dig a bit deeper in this and there was an error in the code uploaded before. Sorry about this.

If we look carefully to the data sheet, we can see that the environmental corrective ratio has to be applied on the Rs/R0 ratio and not on the measurements of the resistances.

[image]

It is fixed in v1.4.1.

closed 2020-07-07 by ostaquet