ostaquet/Arduino-MQ131-driver #20: Questions!

marinajg, 2020-06-12:
Hi Oliver!!! Thank you so much for this library, it's helping me a lot!

1) I noticed that you've changed the sketch and insted of BC337 it's now MOSFET IFR840. But I have the first one, so I want to use it! But in this old scheme, you use a 5,1kOhm resistor on the base to ensure saturation. But, by my calculations, we have to use 2,5kOhm or less, since Ic is 150 mA, than Rb is Hfe(Vbb-Vbe)/Ic = 100(5-1.2)/0.15 = 2500 Ohm. Right? Than I would use a 2,2kOhm resistor to guarantee a higher Ib.

[image]

2) I want to create my own sensitivity curves. To do so, I have to get Rs value, how?! Write this on .CPP?
	float MQ131Class::getRs() {
 		return rS;
	 }
Or just Serial.print(MQ131.readRs());?

3) The rS was calculeted at readRs function. About this one, could you explain to me the math? I just undestand that the analogue pin gives a number between 0 to 1023 and thats why you divide by 1024. But the rest?
	// Compute the voltage on load resistance (for 5V Arduino)
 	float vRL = ((float)valueSensor) / 1024.0 * 5.0;
 	// Compute the resistance of the sensor (for 5V Arduino)
 	float rS = (5.0 / vRL - 1.0) * valueRL;

4) After having built my curves I replace yours in .CPP?

5) I saw that R0 can change a bit with use, so is good to keep thack on it. You recomend to calibrate every time before sampling?

6) Do you have any article published?

I'm doing this as my final work to graduate in environmental engineering, so I don't understand much of eletronics/programing. Sorry for the english too, I'm brasilian. 

Thanks a lot aready!

XOXO, Marina.

ostaquet, 2020-06-17:
See below some answers.

1° I changed the original schema with a MOSFET to have an higher voltage on the heater. Using a BC337 is fine but, even saturated, you don't have 5V on the heater. There is always a drop of voltage in the transistor (Vbe). For the BC337, you end with 4.3V for the heater in ideal conditions. With the MOSFET, you have 5V for the heater, it is better the reach the temperature.

3° The voltage on the analog input is calculated by using the analog to digital from the Arduino. The analog input returns a value between 0 and 1023 which is corresponding the 0V to 5V. So, voltage (VRL) = analog value / 1024 x 5V.

The second part needs a bit more calculation...

The Rs and the RL in the circuit forms a [voltage divider](https://en.wikipedia.org/wiki/Voltage_divider). The concept of a voltage divider is by putting two resistors is series. The VRL measured on the Arduino analog pin is the final voltage coming from the voltage divider (usually called Vout). We know also the Vin (5V).

Vout = Vin * (R2 / (R1 + R2))

Where:
 - Vout = VRL
 - Vin = 5V
 - R2 = RL
 - R1 = Rs

We try to find Rs (so, R1)...

Vout = Vin * (R2 / (R1 + R2))
Vout / Vin = R2 / (R1 + R2)
Vin / Vout = (R1 + R2) / R2
(Vin * R2) / Vout = (R1 + R2)
((Vin * R2) - R2) / Vout = R1
R1 = ((Vin * R2) - R2) / Vout
R1 = ((Vin / Vout) - 1) * R2

So... Rs = ((5V / VRL) - 1) * RL

4° R0 can varies of course. According to some readings on the internet, the calibration process should be done every 3 months but I don't have specific experience on this. Calibration should be in clean fresh air, you should not do it at every sample.

5° No articles published (it was already a long time ago and not about this discipline ;-) ).

closed 2020-06-17 by ostaquet

marinajg, 2020-06-18:
Hi Oliver, again, thank you veeeery much for doing this! I wish you the best!

The last one! 

Could you have chosen "any" resistance instead of 5.1kOhm, since the Hfe ranges from 100 to 630? (respecting the 40mA Arduino limit and knowing that the sensor needs 150mA)

I understand that the current is not relevant as the voltage for the heater...

Thanks!!!

ostaquet, 2020-06-18:
Hi Marina,

You can choose any resistor to control the transistor. I'm use to use the 5.1kOhms to pilot my NPN. The goal is to saturate the transistor. The voltage on the heater is key, that's why I recommend using a MOSFET.

KR