ostaquet/Arduino-MQ131-driver #21: About the heater.

ferreiradiego, 2020-06-14:
Hello, 

You are using the functions startHeater() and stopHeater() to control the sensor heating.
Can I keep the heating all the time? Because your code only starts to heat up when you measure the resistance and then stops heating and so it does a cycle.

juanboque, 2020-06-15:
Hi,

Sorry I´m still not using anything. Once I start I will let you know. This sensor is a bit unclear for me yet.

ostaquet, 2020-06-18:
The method startHeater() and stopHeater() are only present to control the heater. 

It is not a good idea to keep the heater powered all the time because the sensor is semiconductor gas sensor. Basically, the Rs (resistance of the sensor) is changing depending on the gas molecules burned at the surface.

If you keep the heater on all the time, you will never measure anything because the gas molecule will be burned continuously. To have a reliable reading, you have to wait between two samples that the heater is cold again.

closed 2020-06-18 by ostaquet