ostaquet/Arduino-MQ131-driver #29: Derivation of R0

chazbevo, 2020-07-10:
Can you please elaborate on the calculation of R0 - Resistance in fresh air?  Is it derived through an iterative process?  It appears that the R0 comes from a "lastRSvalue" in the program.  Literature on MQ sensors say that the R0 value is calculated from the RS and I'm not quite following it.

MQ131.h shows a default value of  235 ohms, which if I understand correctly, applies to your sensor and would be different for others.  When I calibrate for my sensor, can I use the R0 value and set it as my default in the sketch with "MQ131.setR0()" in the setup?  

Given that I have a high-concentration sensor, can I adjust the R0 values so that the PPM readings are closer to -0- ?

To display debugging, what value do I place after the RL value in  MQ131.begin(2,A0, HIGH_CONCENTRATION, 1000000, **??)**

Thank you.

raphael-bmec-co, 2020-07-16:
I have a follow up question to this. I have been in contact with the sensor manufacturer. They have said that the response curves vary from sensor to sensor for the high-concentration and that curves would need to be gathered for each sensor. Have you by any chance tested multiple sensors? Do you have any idea by what amount the curves vary between sensors?

ferreiradiego, 2020-07-16:
How to obtain the value of R0.

@ostaquet, could you provide or elaborate a process to obtain R0 (the resistance in fresh air)?

The procedure we are doing is as follow:

1. Putting the MQ131 in a closed container with synthetic air (20% oxygen and nitrogen 80%) to simulate clean air.
2. Meanwhile the sensor is in clean air the `calibrate()` function is working to see how the resistence is changing during the calibration and heating time.
3. After the 48h the sensor is heated and then we can get the value of R0.

Do you think this procedure is correct or similar to what you are doing?

We are learning how the sensor work, if anyone have a procedure to provide, we appreciate it.

toze3, 2020-07-17:
This is an important question, because I don't think at all, that the R0 of the High/low sensing ozone  of 235/1917 ohm, and them use a RL of 10kohm (High) and 1Mohm (LOW), simple don´t make sense. I calculated the R0 and the LOW sensor should have around 250Kohm and the HIGH version around 2.500 ohm.

I think there is one more in the market, by ETC (and can be found on sainsmart module) that is planned to work between 10-2000 ppb, and i found that the R0 should be around 12.000 ohms.

I now have a Aeroqual sensor and will try to figure out the curve, just waiting for the ESP or arduino to arrive to compare all readings at the same time.

ostaquet, 2020-07-19:
A lot of questions here :-O
I should consider adding a full explanation on how the sensor is working :-)

First, it is important to understand **how the sensor is working**.

The sensor MQ131 is a semiconductor gas sensor. It means that the resistance of the sensor evolves depending on the quantity of O3 particles that are measured in the air. In order to measure those ozone particles, the heater burns the air and the sensor detects gases by a chemical reaction when the gas comes in direct contact with the sensor.

In a very simplified way, you burn the air and see the remaining of the burning to measure the quantity of gas. The heating time depends on your sensor, that's why it is part of the calibration process.

The **burn-in time of 48h is important** because it burns all remaining particles from the factory process. The remaining particles of the factory process will obviously influence your measures and the burn-in time will stabilize the chemical reaction on the sensor.

In a simplified way, the burn-in time cleanup your sensor before using it ;-)

After the burn-in time, you are able to **measure the resistance of the sensor**. This resistance evolves depending on the chemical reaction. In order to measure it, we use a load resistance in a setup of a voltage divider.

The Rs (resistance of the sensor) and the RL (load resistance) in the circuit are the voltage divider. The concept of a voltage divider is by putting two resistors is series. The VRL measured on the Arduino analog pin is the final voltage coming from the voltage divider (usually called Vout). We also know the Vin (5V).

With a quick calculation (described in issue #20), we arrive at `Rs = ((5V / Vout) - 1) * RL`

We are now able to measure the resistance of the sensor. 👍  The value of RL is not critical as far as you are using a significantly high resistance (10KOhms, 1MOhms...) to be sure that the Vout is influenced significantly by the Rs.

When all this said, we are coming at the core question of **how to measure R0**.

There are 2 parameters set during the calibration process: the R0 and the heating time.

Regarding the R0, it is quite simple... R0 is just the name of the Rs in the clean air. ;-)
The heating time is the time required by heating the sensor to have a stable measure. It is again related to the chemical nature of the sensor. When the heater is on, there are chemical reactions of the gases burning on the sensor which makes _noise_ to measure the O3 concentration. So we have to wait a bit before having a stable measure.

Those parameters are different for each sensor coming out of the factory because of the chemical nature of the sensor. That's why it is critical to make the calibration process.

The calibration process has to be performed when the burn-in time is done (to avoid the influence of the residue from the factory process) in clean fresh air. 

If you want to have a very precise measure, you should use _synthetic air_ coming from bottles (oxygen, azote or a mix), but if you are living in a country where the air quality is quite OK, just put the sensor outside on a Sunday morning. Look at the official air quality measures in your area to be sure there is no high O3 concentration when you're doing your calibration.

When you have your calibration done, you can finally **measure the concentration of O3**.

The sampling process consists of reading the resistance of the sensor after the heating time defined during the calibration to be sure that we have a stable measure.

Again, the sensor resistance (Rs) is related to your sensor and that's why the final concentration is based on a **ratio between the R0 and Rs** (as you can read in the data sheets). As R0 and Rs have been sampled on the same sensor, the chemical variability disappears and the ratio can be used with a formula to find our ppm/ppb concentration (see [sensitivity curves in Excel](https://github.com/ostaquet/Arduino-MQ131-driver/blob/master/extras/datasheet/Sensitivity_curves.xlsx))

ostaquet, 2020-07-19:
@chazbevo To activate the debugging, you have to pass the stream on which you want to see the debug as parameter in the `begin()` function.

Example to output on Serial:
```
MQ131.begin(2,A0, HIGH_CONCENTRATION, 1000000, (Stream *)&Serial)
```

closed 2020-07-19 by ostaquet

raphael-bmec-co, 2020-07-19:
@ostaquet do you have an opinion on my comment that the manufacturer has said the sensitivity curves will be different for each sensor?

ostaquet, 2020-07-19:
@raphael-bmec-co The sensitivity curves are different per sensor due to the chemical reaction. That's why the concentration is measured by comparaison between clean fresh air and ozone air with the ratio of Rs/R0.

raphael-bmec-co, 2020-07-20:
The implication from the manufacturer is that this curve of Rs/R0 varies across sensors:

"
Dear Sir,
The curve in the manual is from one sensor, but every one is different.
"

This is for the high range sensor. They have specifically said it should not be used for measurements without generating curves for each sensor. Have you gathered any data to the contrary or observed some sort of repeatability?

ostaquet, 2020-07-27:
@raphael-bmec-co I didn't make tests with multiples sensors. However, the equation designed from the curves of the data sheet gives pretty close results to the results coming from [the irCELine in Belgium](https://www.irceline.be/en/air-quality/measurements/ozone). irCELine is the official monitoring entity from the government in Belgium. I guess they use calibrated sensors :-)

raphael-bmec-co, 2020-07-28:
@ostaquet, thank you for the feedback. The low concentration sensor is quite repeatable - I am assuming this is what you are working with as you are measuring atmospheric Ozone levels? I am referring to the high concentration sensor 10-1000 ppm. I have the sensor in hand now so I will do some testing in due course and provide some feedback.

Thank you.

ostaquet, 2020-07-29:
@raphael-bmec-co Indeed, I’m talking about the low concentration sensor. I have both but I didn’t compare the results of the high concentration sensor with another data source.