ostaquet/Arduino-MQ131-driver #30: How to obtain the value of R0?

ostaquet, 2020-07-19:
How to obtain the value of R0.

@ostaquet, could you provide or elaborate a process to obtain R0 (the resistance in fresh air)?

The procedure we are doing is as follow:

1. Putting the MQ131 in a closed container with synthetic air (20% oxygen and nitrogen 80%) to simulate clean air.
2. Meanwhile the sensor is in clean air the `calibrate()` function is working to see how the resistence is changing during the calibration and heating time.
3. After the 48h the sensor is heated and then we can get the value of R0.

Do you think this procedure is correct or similar to what you are doing?

We are learning how the sensor work, if anyone have a procedure to provide, we appreciate it.

_Originally posted by @ferreiradiego in https://github.com/ostaquet/Arduino-MQ131-driver/issues/29#issuecomment-659471309_

ostaquet, 2020-07-19:
The procedure is:

1. Heat the sensor for 48 hours (should be done only once when you receive a new sensor from the factory).
2. Putting the MQ131 in a closed container with synthetic air (20% oxygen and nitrogen 80%) to simulate clean air.
3° Start the calibration process.
4° Reuse the results of R0 and heating time from the calibration process as base for your sensor (see #17).

closed 2020-07-19 by ostaquet

LucasCanete, 2024-02-11:
Hello I am having trouble obtaining the value of R0. I put the sensor in a close enclosure and have waited hours but when the library is running MQ131.calibrate() the value of RS and therefore R0 never stabilizes.
Any idea how long this calibration should take?

ostaquet, 2024-02-13:
Did you follow the procedure explained above?

KR,
Olivier