ostaquet/Arduino-MQ131-driver #31: when I apply ozone to the MQ131 sensor instead of increasing the value it drops

acavolina, 2020-09-25:
some help ?

ostaquet, 2020-09-25:
Can you elaborate a bit? :-)

Can you share the wiring, the test protocol and the program?

acavolina, 2020-09-27:
hi, change the sensor (high concentration) and work good the new one, I think was damaged for high temp soldering, the problem now are with the zero level ozone, using RL 1M resistor or 10K the zero are in 18 PPM or 16 PPM, when apply ozone at 15 ppm rise to 24. Now leave the sensor in heating process with 5V directly for 48 hrs to make test again

acavolina, 2020-09-27:
MQ131.setEnv(27,65);    // temp 27o RH = 65%
  lcd.init();             //initialize lcd screen
  lcd.backlight();        // turn on the backlight
  lcd.setCursor(0,0);     // tell the screen to write on the top row
  lcd.begin(16,2);
  pinMode(6, OUTPUT);     // pin 6 rele
  pinMode(5, OUTPUT);     // pin 5 buzzer
  pinMode(3, INPUT_PULLUP);      // pin 3 SELEC
  pinMode(4, INPUT_PULLUP);      // pin 4 ENTER
  Serial.begin(115200);

  // Init the sensor
  // - Heater control on pin 2
  // - Sensor analog read on pin A0
  // - Model LOW_CONCENTRATION
  // - Load resistance RL of 1MOhms (1000000 Ohms)
 //MQ131.begin(2,A0, LOW_CONCENTRATION, 1000000);  
  MQ131.begin(2,A0, HIGH_CONCENTRATION, 1110000); programa:
  lcd.clear(); 
  lcd.setCursor(0,0); 
  lcd.print("Calibracion...");
  
  MQ131.calibrate();
  lcd.clear();
  lcd.setCursor(0,0);
  lcd.print("R0=");
  lcd.print(MQ131.getR0());
  lcd.print(" Ohms");
  lcd.setCursor(0,1);
  lcd.print("calentar en=");
  lcd.print(MQ131.getTimeToRead());
  lcd.print("s");
  lcd.clear();
   delay(5900);
}

ostaquet, 2020-09-28:
The high concentration version of the MQ131 is designed for the high concentration of ozone. It means that the low ozone concentration will produce erratic values.

According to the datasheet, the minimal concentration detected is 10ppm. Knowing that it is a chemical sensor, you will have an high variability at such a low concentration.

I’m not surprized that you cannot reach the 0ppm with the high concentration sensor as it is not designed for this. Try again with high ozone concentration (> 100 ppm) after calibration & burn-in time to validate your results.

Kesavaram, 2020-09-28:
I had the same issue. I suspect your sensor might be a SnO2 sensor where the resistance of the sensor decreases with an increase in O3 concentration.

ostaquet, 2020-11-01:
@acavolina It seems that @Kesavaram is right. He added the code to support the SnO2 version (release 1.5.0).

Can you try your low concentration sensor with the model `SN_O2_LOW_CONCENTRATION` ?

acavolina, 2020-11-02:
thanks I will buy it and try it

closed 2020-11-05 by ostaquet