ostaquet/Arduino-MQ131-driver #34: the time required to heat up the resistance

ChanGarciaV, 2020-10-16:
Hi, I have a question, what happens if I want to take measures quickly, can I turn on the resistor permanently? If I can do that what are the issue with turn on the resistor heater permanently?

ostaquet, 2020-11-01:
Hi @ChanGarciaV,

It is not a good idea to keep the heater powered all the time because the sensor is semiconductor gas sensor. Basically, the Rs (resistance of the sensor) is changing depending on the gas molecules burned at the surface.

If you keep the heater on all the time, you will never measure anything because the gas molecule will be burned continuously. To have a reliable reading, you have to wait between two samples that the heater is cold again.

KR,
Olivier

closed 2020-11-01 by ostaquet

lackdaz, 2022-08-02:
Thanks for the clear explanation, I was also wondering why you were powering down the heater when the preheat time was recommended to be 48hr. This is indeed very misleading

ostaquet, 2022-08-07:
@lackdaz The pre-heat (burn-in) is a one-shot operation to perform. It is required to burn the chemicals that are on the sensor when it comes out of factory.

The heating time (at each measurement) is another story. It is required to generate the chemical reaction of burning particles  of gas. Check out https://en.wikipedia.org/wiki/Gas_detector#Semiconductor for more info about how it behaves theoretically. So, the heating at the measurement must be stopped to perform the measurement ;-).