ostaquet/Arduino-MQ131-driver #36: Changing 5V to 3.3V on the mq131.cpp readRS()function

alf-esp32, 2020-11-05:
I'm using the mq131 Ozone sensor on an ESP32 (rather than Arduino) which internal operating voltage (and max ADC input) is 3.3V
Hence I'm feeding the sensor with 3.3V instead of 5V.

Would it work if I change the library mq131.cpp and just change 5.0 (for 5V Arduino) to 3.3 (for 3.3V ESP32)?

I mean at the readRS() function
float MQ131Class::readRs() {
// Read the value
uint16_t valueSensor = analogRead(pinSensor);
// Compute the voltage on load resistance (for 5V Arduino)
float vRL = ((float)valueSensor) / 1024.0 * 5.0; // would it work if I place 3.3 instead of 5.0?
// Compute the resistance of the sensor (for 5V Arduino)
if(!vRL) return 0.0f; //division by zero prevention
float rS = (5.0 / vRL - 1.0) * valueRL; //same here; 3.3 instead of 5.0?
return rS;
}

By the way, I'm using the 1.4.4 version since 1.5 version does not compile for me (some undefined variables)

ostaquet, 2020-11-05:
Hi @alf-esp32,

Thanks for the feedback on the 1.5.0. I merge the pull request without checking in details... It is my fault and it is now fixed in 1.5.1.

There are two power lines to the MQ131 sensor. The heater circuit needs 5V to be sure that you're burning the gas particle and have the right chemical reaction. The sensor circuit could be powered by a 3.3V line (instead of the 5V line) as the sensor is acting as a resistor that vary depending on the gas concentration. So, by changing the 5V into 3.3V should work ;-)

closed 2020-11-07 by ostaquet

Olvera-Orozco-Miguel-Angel, 2023-09-07:
Hello,

I tested the code you shared with Arduino (I mean this https://github.com/ostaquet/Arduino-MQ131-driver) , and it works perfectly. However, I need to use an ESP32 now, and I'm not sure how to connect the MQ131 sensor to the ESP32. Could you please provide some guidance?

Thank you!