ostaquet/Arduino-MQ131-driver #37: Sensitivity curves

morfeus2, 2020-12-07:
Hello, thanks for the project, it's very interesting.
I would like to ask you: where did you get the calibration values presented in "Sensitivity curves" spreadsheet?

dzvid, 2020-12-07:
@morfeus2 You mean the values in the tables?

morfeus2, 2020-12-07:
Yes, exactly Rs/R0 vs ppm, Influence of Temperature/Humidity etc...
Were they measured by hand from the graphs in the official datasheet or something better?

dzvid, 2020-12-09:
It was measured from datasheet curves, getting some points in the curve and applying linear regression to get sensor equations. 
As said here: #29

ostaquet, 2020-12-09:
Indeed, as @dzvid wrote, the data points are coming from the data sheet and I applied a regression to get the best sensor equation.

closed 2020-12-09 by ostaquet