ostaquet/Arduino-MQ131-driver #38: Time between readings (Heater on, Heater off)

SciWax, 2021-01-14:
Hello ostaquet,

I would like to ask you about your experience. In this thread you said, it's advised to let the MQ131 cool down inbetween readings:

https://github.com/ostaquet/Arduino-MQ131-driver/issues/34#issuecomment-720098987

At the moment, I'm measuring other variables every 10 seconds. 

In your experience, what is the highest measurement frequency I can get with the MQ131? How long should the Heater be off, before I can turn the Heater on again to take a measurement? How long has the Heating period to be, before I take a measurement? How long are your cooling periods and heating periods. Do you take a single measurement after the heater is on for a certain amount of time? Thank you for any help!

Kind Regards,

SciWax

ostaquet, 2021-01-17:
I didn't make any experiments on this aspect because my usage is to sample the data every 15 minutes. It is plenty of time to let the sensor cold down.

In order to cold down the sensor, it is obviously linked to the environment. If it is outside by -10°C, it will cool much faster than on the beach at 35°C.

There is a first information in the datasheet. According to the response curves mentioned in the datasheet, we should expect at least 30 to 40 seconds between two samples but there is no information about the environment and the conditions of this experiment.

Depending on your usage and conditions, you should test how the sensor cools down between two samples.

SciWax, 2021-01-18:
> I didn't make any experiments on this aspect because my usage is to sample the data every 15 minutes. It is plenty of time to let the sensor cold down.
>
> In order to cold down the sensor, it is [...]

Thank you for the advise! So there is plenty to experiment with. Personally, I'm not sure, if the measurement intervall is 40 seconds in the response curve. When you look at the curve for the low concentration sensor, you can see small little dots for instance between 0 and 40 seconds. These dots look like individual measurements, when you zoom in, which makes me think, that they measured at a way higher frequency. At some point, the sensor just seems to get saturated, because the concentration of the test gas stays at 200 ppb?  

In your experience, after the cooling down phase, how long should the Heater be on before I take measurements? I was thinking about heating for 1 Minute before actually taking any measurements to be safe even during really cold phases. 
Kind regards,

SciWax

ostaquet, 2021-03-13:
Do you have a chart with the different measurements?
What is the temperature of the environment around the sensor?

closed 2021-05-18 by ostaquet