ostaquet/Arduino-MQ131-driver #39: CAN WE USE MQ-131 WITH MODULE THAT HAS 4 PINS (Vcc, gnd, A0,D0) ? INSTEAD OF USING THE SENSOR'S ALL 6 PINS?

lailomah, 2021-01-19:
Thankyou for an organized tutorial about MQ-131 I have few questions if someone could answer it will be so helpful.

1: Instead of using all 6 pins of heater and sensors Can we use this Sensor with the module that has 4 pins (VCC,GND, A0. D0) ?

2: How the library is calculating PPM? is it converting the analogue values into ppm?

ostaquet, 2021-03-13:
1. The sensors have always 6 pins but only 4 are really used. I don't have the documentation of your module. So, it is difficult to know the parameters of the breakout module. Do you have some doc?

2. The result of the sensor is a resistance ratio. The resistance ratio is computed based on the sensitivity curves of the sensor as defined in the datasheet. The formula are in the Excel sheet (https://github.com/ostaquet/Arduino-MQ131-driver/blob/master/extras/datasheet/Sensitivity_curves.xlsx).

closed 2021-05-18 by ostaquet

damico, 2021-09-21:
I  believe that **lailomah** is talking about this board:

[image]

Which can be understood by this circuit diagram:

[image]

Where the R2 is 1KOhm.

Best Regards,

Jose Damico