ostaquet/Arduino-MQ131-driver #40: Two sensors MQ131: LOW and HIGH concentration in project

Chris722, 2021-01-23:
Hello ostaquet,

I'd like to connect two MQ131 sensors to Arduino UNO: first - LOW_CONCENTRATION to A0 input and second HIGH_CONCENTRATION to A1 input.
How should I declare them in the program? 

Best regards,
KAD

ostaquet, 2021-03-13:
As you can see in the header file, the sensor is created through the instance of `MQ131Class`.

In order to have multiple instance of the driver, you just have to instantiate the class more than once.

Per example:
```
MQ131Class lowSensor = new MQ131Class();
MQ131Class highSensor = new MQ131Class();

lowSensor->begin(...);
highSensor->begin(...);
[...]
```

closed 2021-03-13 by ostaquet

tbmart16, 2021-04-14:
Hi ostaquet,

I have a similar goal of using both a high concentration and low concentration sensor in my program. I see your response above and I tried to implement it into my Arduino code, but I get an error. Should I put your example above before the void setup() or inside of the void setup()?

Chris722, 2021-04-15:
Hi tbmart16,

This code works for me:

#include <MQ131.h>

MQ131Class highSensor = new MQ131Class(8000);   # Load resistance RL of 8kOhms (8000 Ohms)
MQ131Class lowSensor = new MQ131Class(1000000); # Looad resistance RL of 1MOhms (1000000 Ohms)

void setup() {
  Serial.begin(115200);
  
  highSensor.begin(2,A0, HIGH_CONCENTRATION, 8000, (Stream *)&Serial); 
[...]