ostaquet/Arduino-MQ131-driver #50: MQ131 and ESP32

dimitris1965, 2022-04-18:
Dear friend.
Can I use this library with esp32?

ostaquet, 2022-05-01:
Hi @dimitris1965 
It should work. However, some adjustment maybe needed to inform the watchdog that the process is still running. 
Some `sleep()` functions could not be appreciated by the watchdog (typically during calibration, warm-up or waiting time for the sample).
As I don't have ESP32 here, I cannot test it.
Let me know if you have issue and we can look at it.
KR

dimitris1965, 2022-05-01:
My ESP32 is the "fire beetle" ESP32 WROOM 32E.
 I don't use any sleep() function.
The number MQ131_DEFAULT_STABLE_CYCLE (15) is too big. It takes over an hour to reach 5, I don't know why.
Another issue is the MOSFET. I must use a MOSFET that can be controlled by 3.3v (so far I use a relay). I already order an IRLZ44N.
I apply to sensor and heater 5V and read it via voltage divider. I have change to library the 1023 to 4095.
The problem for now is the number MQ131_DEFAULT_STABLE_CYCLE.

lackdaz, 2022-08-02:
hey just wanted to tag on this issue, I'm using the ESP8266 that also has a 3.3v ADC, what do I set the load resistance to?

ostaquet, 2022-08-07:
@lackdaz Load resistance with 3.3V is answered in #54

closed 2023-05-05 by ostaquet