ostaquet/Arduino-MQ131-driver #55: Unusual Sensor Reading

NXTkoji, 2022-08-08:
Hello. I hope you can help me out again. 

I have preheated the MQ-131 high concentration for 36 hours.
My wiring connections are almost same as yours, except that I have an external power supply.

[image]

The value from the sensor is unusually high, and was wondering if I did something wrong in calibration. Can you please help out?

Following is the result on serial monitor:
 *  Executing task in folder MQ131test: platformio device monitor --environment uno 

--- Terminal on /dev/cu.usbserial-1130 | 9600 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
Calibration in progress...
Calibration parameters
R0 = 977.52 Ohms
Time to heat = 17 s
Sampling...
Concentration O3 : 12.34 ppm
Concentration O3 : 12343.99 ppb
Concentration O3 : 26.09 mg/m3
Concentration O3 : 26089.09 ug/m3
Sampling...
Concentration O3 : 12.34 ppm
Concentration O3 : 12343.99 ppb
Concentration O3 : 26.09 mg/m3
Concentration O3 : 26089.09 ug/m3
Sampling...

My code is as follows:
`#include <Arduino.h>
#include <MQ131.h>

int temperature=26; //degrees celcius
int humidity=50;  // percent

void setup() {
 //Serial.begin(74880);
 Serial.begin(9600);

  // Init the sensor
  // - Heater control on pin 2 (D4 if MiniD1)
  // - Sensor analog read on pin A0
  // - Model HIGH_CONCENTRATION
  // - Load resistance RL of 1MOhms (1000000 Ohms)
  MQ131.begin(2,A0, HIGH_CONCENTRATION, 1000000);  

  Serial.println("Calibration in progress...");
  MQ131.calibrate();

  Serial.println("Calibration parameters");
  Serial.print("R0 = ");
  Serial.print(MQ131.getR0());
  Serial.println(" Ohms");
  Serial.print("Time to heat = ");
  Serial.print(MQ131.getTimeToRead());
  Serial.println(" s");

  MQ131.setR0(MQ131.getR0());
  MQ131.setTimeToRead(MQ131.getTimeToRead());
}

void loop() {
  delay(60000);

  MQ131.setEnv(temperature, humidity);
  
  Serial.println("Sampling...");
  MQ131.sample();

  Serial.print("Concentration O3 : ");
  Serial.print(MQ131.getO3(PPM));
  Serial.println(" ppm");
  Serial.print("Concentration O3 : ");
  Serial.print(MQ131.getO3(PPB));
  Serial.println(" ppb");
  Serial.print("Concentration O3 : ");
  Serial.print(MQ131.getO3(MG_M3));
  Serial.println(" mg/m3");
  Serial.print("Concentration O3 : ");
  Serial.print(MQ131.getO3(UG_M3));
  Serial.println(" ug/m3");

}
/*******************************************************************************
 * Sample the ozone concentration every 60 seconds
 * 
 * Example code base on high concentration sensor (metal)
 * and load resistance of 1MOhms
 * 
 * Schematics and details available on https://github.com/ostaquet/Arduino-MQ131-driver
 ******************************************************************************
 * MIT License
 *
 * Copyright (c) 2018 Olivier Staquet
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 * 
 * The above copyright notice and this permission notice shall be included in all
 * copies or substantial portions of the Software.
 * 
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
 * SOFTWARE.
 *******************************************************************************/`

NXTkoji, 2022-08-11:
I tried with another MQ-131 (high concentration), brand new, preheated sensor, but the readings were the same as the previous test. Following is the message from the serial monitor.

Calibration in progress...
Calibration parameters
R0 = 977.52 Ohms
Time to heat = 17 s
Sampling...
Concentration O3 : 12.34 ppm
Concentration O3 : 12343.99 ppb
Concentration O3 : 26.09 mg/m3
Concentration O3 : 26089.09 ug/m3
Sampling...
Concentration O3 : 12.34 ppm
Concentration O3 : 12343.99 ppb
Concentration O3 : 26.09 mg/m3
Concentration O3 : 26089.09 ug/m3

NXTkoji, 2022-08-12:
Found out that you have specified to change code in your library for high concentration.
' 		case HIGH_CONCENTRATION :
 			// Use the equation to compute the O3 concentration in ppm
 			
      // Compute the ratio Rs/R0 and apply the environmental correction
      ratio = lastValueRs / valueR0 * getEnvCorrectRatio();
      // R^2 = 0.9900
      // Use this if you are monitoring low concentration of O3 (air quality project)
      return convert(8.1399 * pow(ratio, 2.3297), PPM, unit);
      
      // R^2 = 0.9985 but nearly impossible to have 0ppm
      // Use this if you are constantly monitoring high concentration of O3
      // return convert((8.37768358 * pow(ratio, 2.30375446) - 8.37768358), PPM, unit);'

Followed the instruction and now I have a better reading.

Calibration parameters
R0 = 977.52 Ohms
Time to heat = 17 s
Sampling...
Concentration O3 : 4.27 ppm
Concentration O3 : 4268.12 ppb
Concentration O3 : 9.02 mg/m3
Concentration O3 : 9020.70 ug/m3

ostaquet, 2022-09-12:
Hi @NXTkoji,

The high concentration sensor is manufactured with a detection range from 10ppm to 1000ppm (and not ppb ;-) ).

If you are performing measurements in clean air, I doubt you will reach 10ppm in real measurements.

As an example, the ozone in Belgium is about 107 ug/m3 (https://www.irceline.be/fr/qualite-de-lair/mesures/ozone). It means around 54 ppb (about 0.055 ppm). We are far below the minimum of the sensor.

The value you got in clean air is just a mathematical interpretation.

The high concentration sensor is really designed for the, as named :-), the high concentration sensing.

I hope it helps ;-)

closed 2022-09-12 by ostaquet