ostaquet/Arduino-MQ131-driver #56: Low Concentration Qualitative Detection MQ131 with ESP32

JoshDrck, 2022-08-31:
Hello, I need your help.
I need integrate a MQ131 sensor With Esp32, but I am working with this model: 
[image]
And I dont know how add the MOSFET in the conection.
Can I still use a library? And what would happen with that connection to the voltage divider?
I'm a amateur in this area.
Sorry for my English.
Thank.

ostaquet, 2022-09-12:
Do you have the datasheet for this breakout board?

I see a potentiometer, probably to setup the load resistance (RL) but without details, it is complex to answer.

ostaquet, 2023-05-05:
Closed as won't fix, missing info.

closed 2023-05-05 by ostaquet